espressif/esp-brookesia #2: Driff lcd when rotate  with esp32_s3_lcd_ev_board

isthaison, 2024-08-21:
[image]
driff lcd when rotate 
[image]
[image]

Lzw655, 2024-08-22:
Hi @isthaison,

Sorry, this is likely an RGB screen tearing issue. You can set `BSP_LCD_RGB_REFRESH_MODE` to `BSP_LCD_RGB_BOUNCE_BUFFER_MODE` to resolve it. By the way, you should disable the anti-tearing feature `BSP_DISPLAY_LVGL_AVOID_TEAR` before using LVGL rotation. Otherwise, it won't take effect.

Additionally, there are display issues when using LVGL software rotation with esp-ui, so it's currently recommended not to use LVGL rotation.

isthaison, 2024-08-22:
I got it, thanks you

closed 2024-08-22 by isthaison

chegewara, 2025-02-17:
Hi,
yes, it would be nice to have rotate working.
I just bought waveshare p4 nano with 10" display 800x1280 and i would prefer to not use it in portrait mode.