espressif/esp-brookesia #5: Phone Resolution Support

Lzw655, 2024-08-22:
If there is a resolution in the list below that is close to what you need, please make minor adjustments based on it. Otherwise, please suggest the desired resolution in the comments, and we will do our best to accommodate your request.

> [!WARNING]
> Since the current stylesheet interface of esp-ui is not yet stable, we do not recommend deep customization at this time. It is expected to stabilize after version 1, at which point you can use the PC simulator or proceed with deep customization in your development. 

The following is a list of resolutions to be implemented, arranged by priority:

|  Resolution  |                      Status                       | link |
| ------------ | ------------------------------------------------- | ---- |
|  480 x 480   | [image] | [Github](https://github.com/esp-arduino-libs/esp-ui-phone_480_480_stylesheet) / [Espressif](https://components.espressif.com/components/espressif/esp-ui-phone_480_480_stylesheet) |
|  800 x 480   | [image] | [Github](https://github.com/esp-arduino-libs/esp-ui-phone_800_480_stylesheet) / [Espressif](https://components.espressif.com/components/espressif/esp-ui-phone_800_480_stylesheet) |
| 1024 x 600   | [image] | [Github](https://github.com/esp-arduino-libs/esp-ui-phone_1024_600_stylesheet) / [Espressif](https://components.espressif.com/components/espressif/esp-ui-phone_1024_600_stylesheet) |
|  320 x 240   | [image]  | / |
| 1280 x 800   | [image]  | / |
|  320 x 480   | [image]  | / |

Y1hsiaochunnn, 2024-09-07:
Hi @Lzw655 ,
  I was wondering if there are plans to add 1080x1920 resolution support?

Lzw655, 2024-09-10:
Hi @Y1hsiaochunnn,

Sorry, as this repository is going to be renamed and the stylesheet-related APIs will undergo significant changes, we are temporarily suspending support for new resolutions. Thank you for your understanding.

sukesh-ak, 2025-05-07:
> Hi [@Y1hsiaochunnn](https://github.com/Y1hsiaochunnn),
> 
> Sorry, as this repository is going to be renamed and the stylesheet-related APIs will undergo significant changes, we are temporarily suspending support for new resolutions. Thank you for your understanding.

Any update on the name changes etc mentioned earlier?

Lzw655, 2025-05-08:
Hi @sukesh-ak,

The current Phone interface will no longer undergo major changes. We will only update it in the subsequent new system UI.

Lzw655, 2025-08-26:
Hi @ronger-x,

Since you have this requirement, I can add 480 x 800 to the plan list, and it is expected to be completed next week.

chenguang3312, 2025-09-28:
你好，我这边屏幕是800*1280，默认竖屏模式。但是我需要用1280*800 style显示在800*1280屏幕上请问如何做？ 单纯lvgl示例通过软件旋转可以实现，但是使用brookesia-phone 800*1280style时直接旋转会报错。
`[WARN] [esp_brookesia_core.cpp:46](getDisplaySize): Display is not set, use default display
[ERROR][esp_brookesia_core_home.cpp:128](calibrateCoreObjectSize): Invalid value: 1280, should be in range [1, 
800]
[ERROR][esp_brookesia_core_home.cpp:128](calibrateCoreObjectSize): Invalid Height
[ERROR][esp_brookesia_phone.cpp:169](calibrateScreenSize): Invalid screen size
[ERROR][esp_brookesia_stylesheet_template.hpp:110](addStylesheet): Invalid screen size
[ERROR][esp_brookesia_phone.cpp:102](addStylesheet): Failed to add phone stylesheet
[ERROR][main.cpp:63](app_main): Add phone stylesheet failed`
尝试使用1280*800 style 修改display接口，渲染也存在问题。