espressif/esp-brookesia #15: Status Bar Clock to support display 12hr or 24hr format

rysmith0315, 2024-12-06:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

The status bar widget's clock display should allow for localization.   It should be possible to display a 12 hour format with AM/PM or switch to a 24hr format without AM/PM.  

### Use cases

Support localization to display in the preferred format for a country.  Allows for creating products that meet the expected formats by end users in the target market.

### Alternatives

An alternative would be to use a localization table to control various display elements, such as time, date, currency, temp, etc.

Lzw655, 2024-12-10:
Hi @rysmith0315, thank you for your suggestion! We can add this feature in the next version.

closed 2024-12-23 by Lzw655

rysmith0315, 2025-01-18:
AM/PM is now display but the time is still in 24H format.

Suggested changes:
esp_brookesia_status_bar.hpp:
100: mutable ClockFormat _clock_format;

esp_brookesia_status_bar.cpp
600: _clock_format = format;
664: _clock_format = format;
682:     if (_clock_hour != hour) {
683:        _clock_hour = hour;
684:        if (_clock_format == ClockFormat::FORMAT_12H) {
685:            hour = hour % 12;
686:            if (hour == 0) {
687:                hour = 12;
688:        }
689:     }