espressif/esp-brookesia #21: LVGL 9 support

nicklasb, 2025-02-11:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

LVGL9 has been out a year now, it would be great if esp-brookesia moved to it before it has a lot of users.

I see no reason for a really new library like esp-brookesia to move to an old LVGL version as it is unlikely that old project could move directly to it anyway.

(Note also that Squareline support LVGL 9, and also that LVGL are building an open generator themselves)

### Use cases

People not having to move from LVGL 8 to 9 as one of the first things they will have to do if they start using esp-brookesia in a year when LVGL8 naturally goes out of focus for the LVGL project.

Lzw655, 2025-02-11:
Hi @nicklasb,

Thank you very much for your feedback! I completely agree with your point of view. The migration from LVGL 8 to 9 should not be too difficult, with the main work likely involving re-exporting images and other resource files. It looks like it's time to put this into the plan.  

Additionally, I’ve been considering implementing a C++ layer for GUI abstraction based on LVGL and enabling a dynamic UI generation approach similar to LVGL 9’s XML-based system. This would better align with modern UI design practices.

espzav, 2025-02-11:
> re-exporting images

Hi @Lzw655,
the LVGL port supports automatic generating images for LVGL8/LVGL9. Here is the example: https://github.com/espressif/esp-bsp/blob/35a10dc3ff6435776e456648af6b915bea1af01f/examples/display/main/CMakeLists.txt#L4

nicklasb, 2025-02-11:
Ok, sounds promising. When will this happen? 
Because I just made my framework LVGL 9 only, so I am kind of incompatible with this library currently.

nicklasb, 2025-02-11:
@Lzw655 So I sat down and put down a few hours to port esp-brookesia it to LVGL 9, and as expected, most was pretty obvious (lv_img_dsc_t, lv_obj_send_event and so on), but after a while I got really into the weeds as I ended up getting strange errors like this, code I risked breaking without being able to understand how to fix it.

```
..../components/esp-brookesia/src/widgets/gesture/esp_brookesia_gesture.cpp: In static member function 'static void ESP_Brookesia_Gesture::onTouchDetectTimerCallback(lv_timer_t*)':
..../components/esp-brookesia/src/widgets/gesture/esp_brookesia_gesture.cpp:502:64: error: invalid use of incomplete type 'struct lv_timer_t'
  502 |     ESP_Brookesia_Gesture *gesture = (ESP_Brookesia_Gesture *)t->user_data;
```
Actually I think I didn't have much left to do had I understood the code, you'd probably be able to do it much quicker.

Lzw655, 2025-02-12:
> > re-exporting images
> 
> Hi [@Lzw655](https://github.com/Lzw655), the LVGL port supports automatic generating images for LVGL8/LVGL9. Here is the example: https://github.com/espressif/esp-bsp/blob/35a10dc3ff6435776e456648af6b915bea1af01f/examples/display/main/CMakeLists.txt#L4

Hi @espzav,

Awesome! This is much more convenient than manually using a web tool to convert images. Thank you for sharing!

Lzw655, 2025-02-12:
> Ok, sounds promising. When will this happen? Because I just made my framework LVGL 9 only, so I am kind of incompatible with this library currently.

I think the earliest will be mid-March.

Lzw655, 2025-02-12:
> [@Lzw655](https://github.com/Lzw655) So I sat down and put down a few hours to port esp-brookesia it to LVGL 9, and as expected, most was pretty obvious (lv_img_dsc_t, lv_obj_send_event and so on), but after a while I got really into the weeds as I ended up getting issues like this, code I risked breaking without being able to understand how to fix it. For example, I realize that the gesture is in the user data, but I feel like I venture into areas where I am unsure of intents makes to many changes where I assume:
> 
> ```
> ..../components/esp-brookesia/src/widgets/gesture/esp_brookesia_gesture.cpp: In static member function 'static void ESP_Brookesia_Gesture::onTouchDetectTimerCallback(lv_timer_t*)':
> ..../components/esp-brookesia/src/widgets/gesture/esp_brookesia_gesture.cpp:502:64: error: invalid use of incomplete type 'struct lv_timer_t'
>   502 |     ESP_Brookesia_Gesture *gesture = (ESP_Brookesia_Gesture *)t->user_data;
> ```
> 
> Actually I think I didn't have much left to do had I understood the code, you'd probably be able to do it much quicker.

You are right, the code structure of esp-brookesia still needs optimization. 

I think the issue in the example you provided is due to LVGL9 hiding many element structures in *lvgl_private.h*, which is not included by default. For example, `_lv_timer_t` is located in *lv_timer_private.h*. I believe including *lvgl_private.h* in the source file should resolve this issue.

yihua-wang, 2025-04-15:
https://github.com/yihua-wang/esp-brookesia
I have made it run on lvgl9.2 and only support the resolution of my board. 
The squareline has been removed.

nicklasb, 2025-05-10:
@yihua-wang @Lzw655 
This is exciting, have completely missed the notifications on this issue, I am going to have a look at the lvgl 9 branch tomorrow, and see if I can get it running.