espressif/esp-brookesia #22: NavBar disappear

chegewara, 2025-02-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Hi,
this components is great (cant wait to see lvgl9 port too), but i have tiny issue with navbar.
When i add it in the main code with
```
phone->getHome().getNavigationBar()->setVisualMode(ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED);
```
and then open some app, or just click on `Recent apps` then navbar is gone.
If i am doing something wrong, then im sorry.

Thanks

### Actual behavior (suspected bug)

-

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

-

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu

### Shell

Bash

Lzw655, 2025-02-14:
Hi @chegewara,

Thank you for trying out this library! No worries and feel free to ask.

Currently, it only supports modifying the navigation bar's display mode statically through a stylesheet. You can refer to the following code:

```c
    ESP_Brookesia_Phone *phone = new ESP_Brookesia_Phone();    
    ...
    // Can replace `ESP_BROOKESIA_PHONE_DEFAULT_DARK_STYLESHEET` with yours
    ESP_Brookesia_PhoneStylesheet_t *stylesheet = new ESP_Brookesia_PhoneStylesheet_t ESP_BROOKESIA_PHONE_DEFAULT_DARK_STYLESHEET();
    stylesheet->home.navigation_bar.visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
    stylesheet->home.recents_screen.navigation_bar_visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
    phone->addStylesheet(stylesheet);
    phone->activateStylesheet(stylesheet);
    phone->begin()
    ...
    phone->begin();
```

This way, you can see the navigation bar on the main page and in the recent apps. Please note that if you are using a custom resolution stylesheet (e.g., `ESP_BROOKESIA_PHONE_720_1280_DARK_STYLESHEET`), changing the navigation bar status may fail to initialize due to stylesheet validation errors. In such cases, you need to adjust the relevant styles based on the error logs. If you need help and cannot resolve the issue, I can customize a stylesheet for you.

Additionally, each app can individually set the navigation bar's display mode. You can refer to the [simple example](https://github.com/espressif/esp-brookesia/blob/024106cfd56c8bb07022f7c6dde6a6e92499b0d9/src/app_examples/phone/simple_conf/src/phone_app_simple_conf.cpp#L12-L15) or [complex example](https://github.com/espressif/esp-brookesia/blob/024106cfd56c8bb07022f7c6dde6a6e92499b0d9/src/app_examples/phone/complex_conf/src/phone_app_complex_conf.cpp#L39) to set `use_navigation_bar` to `true` in your custom app.

chegewara, 2025-02-14:
Hi,
thanks for explanation.
This didnt work for me, but i think i missed one line of code and got some errors (i think i can live w/o it)
```
    ESP_Brookesia_PhoneStylesheet_t *stylesheet = new ESP_Brookesia_PhoneStylesheet_t ESP_BROOKESIA_PHONE_DEFAULT_DARK_STYLESHEET();
    stylesheet->home.navigation_bar.visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
    stylesheet->home.recents_screen.navigation_bar_visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
```
This one i already figured out and works as expected
> Additionally, each app can individually set the navigation bar's display mode. You can refer to the [simple [...]

Thanks

chegewara, 2025-02-14:
Hi @Lzw655 
this code is not working
```
    ESP_Brookesia_PhoneStylesheet_t *stylesheet = new ESP_Brookesia_PhoneStylesheet_t ESP_BROOKESIA_PHONE_DEFAULT_DARK_STYLESHEET();
    stylesheet->home.navigation_bar.visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
    stylesheet->home.recents_screen.navigation_bar_visual_mode = ESP_BROOKESIA_NAVIGATION_BAR_VISUAL_MODE_SHOW_FIXED;
```

here is error log
```
I (2416) app_main: Display ESP-Brookesia phone demo
I (2421) app_main: Using stylesheet (1024x600 Drak)
E (2426) esp-brookesia: [esp_brookesia_core_home.cpp:128](calibrateCoreObjectSize)Invalid value: 420, should be in range [1, 350]
E (2437) esp-brookesia: [esp_brookesia_core_home.cpp:128](calibrateCoreObjectSize)Invalid Height
E (2445) esp-brookesia: [esp_brookesia_recents_screen.cpp:463](calibrateData)Invalid snapshot main size
E (2454) esp-brookesia: [esp_brookesia_phone_home.cpp:395](calibrateData)Calibrate recents_screen data failed
E (2464) esp-brookesia: [esp_brookesia_phone.cpp:156](calibrateStylesheet)Invalid home data
E (2472) esp-brookesia: [esp_brookesia_stylesheet_template.hpp:113](addStylesheet)Invalid stylesheet
E (2481) esp-brookesia: [esp_brookesia_phone.cpp:102](addStylesheet)Failed to add phone stylesheet
E (2490) esp-brookesia: [esp_brookesia_phone.cpp:114](addStylesheet)Failed to add phone stylesheet
E (2498) esp-brookesia: [main.cpp:62](app_main)Add stylesheet failed
```

Thanks

Lzw655, 2025-02-14:
As I mentioned here:

> Please note that if you are using a custom resolution stylesheet (e.g., ESP_BROOKESIA_PHONE_720_1280_DARK_STYLESHEET), changing the navigation bar status may fail to initialize due to stylesheet [...]

closed 2025-02-14 by chegewara