espressif/esp-brookesia #23: smartphone like widget on home screen

chegewara, 2025-02-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

Hi, is there any chance to add smartphone/android like custom widgets on home screen?
What i mean is not just an icon to open app, but widget docked on screen to perform some actions or to display some info, like weather or something.

### Use cases

With 1 button press perform action without opening app, ie:
- enable/disable wifi
- turn on/off light 
- turn on/off display
- restart or deep/light sleep

I see a lot more single button press usage cases.

Thanks

chegewara, 2025-02-13:
Hi team,
I dont want to open new topic, so i will try to ask another question here:

- is it possible to start/open/run brookesia app from code?

Thanks

Lzw655, 2025-02-14:
Hi @chegewara,

Your idea is great! Although this library currently doesn't support adding custom widgets on the home screen, I think it should be possible to make an app icon act as a button that doesn't open a full interface when clicked. Give me some time to try it out. 

As for how to start/stop an app via code, you can refer to the following implementation:

```c
    ...
    // Get app ID when installing
    int id = phone->installApp(app_simple_conf);
    ...
    // Send event to start/stop app
    ESP_Brookesia_CoreAppEventData_t app_data = {
      .id = id,
      .type = ESP_BROOKESIA_CORE_APP_EVENT_TYPE_START, // or ESP_BROOKESIA_CORE_APP_EVENT_TYPE_STOP
    };
    phone->sendAppEvent(&app_data);
```

chegewara, 2025-02-14:
Thank you very much
I will try for sure, because i may have some use cases for it

chegewara, 2025-02-16:
Hi,
maybe its not most elegant way to do it, but for now it works:
```
bool OneShotWidget::run(void)
{
    ESP_BROOKESIA_LOGD("Run");
    // trigger some non-blocking action here
    return false; // return false to not create lvgl screen
}
```

Thanks

Lzw655, 2025-02-18:
> Hi, maybe its not most elegant way to do it, but for now it works:
> 
> ```
> bool OneShotWidget::run(void)
> {
>     ESP_BROOKESIA_LOGD("Run");
>     // trigger some non-blocking action here
>     return false; // return false to not create lvgl screen
> }
> ```
> 
> Thanks

You're right! I also want to try this method. This might be the only feasible method for now. I will try to add this feature in the next version.

chegewara, 2025-02-25:
Hi @Lzw655 
i like brookesia very much and i am playing with it a lot right now. Building app still requires a lot of lvgl codding, but makes my life a lot easier.
I have few ideas how to make it "more friendly" to end user, but i dont know if anything like that is even possible:
- is it possible to make status icon "clickable"
- drag and drop icons with long press, to change order on screen
- make clock clickable, so we could add keyboard and let user to set time (or at least some option to get handle to clock)

Big thank you for that component

Lzw655, 2025-02-26:
Hi @chegewara,

Thank you very much for your support of this library. 

In principle, I believe the new features you mentioned can be implemented. However, as you said, *"Building an app still requires a lot of LVGL coding."* This GUI design approach makes it difficult to be compatible with devices of different resolutions. Therefore, I would like to prioritize implementing the XML-like design approach I mentioned earlier. Of course, I will also gradually implement the features you brought up.