espressif/esp-brookesia #24: Demo Apps

bensuffolk, 2025-02-25:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

In your video showing what can be done there are a few demo apps, like WiFi setup. It would be nice if they were made available.

### Use cases

This would save re writing WiFi set up apps, and give some more insight into app development

Lzw655, 2025-02-26:
Hi @bensuffolk,

Thank you for using this library. We are planning to create a new GitHub repository to store these apps, but we still need some time to organize the code. This work is expected to be completed by March.

neotelsha, 2025-07-07:
hi @Lzw655  Can you tell me where find demo apps repository? or we still need wait?
Thank you !

neotelsha, 2025-07-07:
> Hi [@neotelsha](https://github.com/neotelsha),
> 
> There are some demos in the [apps](https://github.com/espressif/esp-brookesia/tree/master/apps) directory, but most of them are currently only applicable to the `Speaker` system and the `EchoEar` development board. We still need to study how to make them compatible with other systems (such as `Phones`) and development boards.

Thank you very much