espressif/esp-brookesia #42: Use CST820 as Touch device, it is initialized in driver, but could not get LvInputDev

lznzhou33zzz, 2025-08-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

[echoear.zip](https://github.com/user-attachments/files/21746680/echoear.zip)

Use CST820 as Touch device, it is initialized in driver, but could not get LvInputDev.
It cause ESP32S3 reboot.

Hardware is Echoear based board, main differences are use an AMOLED screan, its touch IC is CST820, LCD driver IC is CO5300, both driver code are ported. No SD storage included in this design. 

Main error infor as  below
E (10333) BS:Speaker: [esp_brookesia_speaker_manager.cpp:0226](begin): No touch device is initialized
E (10333) BS:Speaker: [esp_brookesia_speaker.cpp:0089](begin): Failed to begin manager
E (10334) System: [system.cpp:0127](system_init): Begin failed

assert failed: void app_main() main.cpp:32 (system_init() && "Initialize system failed")

Log file and modified bsp code are in zip file.

Lzw655, 2025-08-14:
Hi @lznzhou33zzz,

From your code, I see that `lv_display_create()` is called in `bsp_display_new()` to create the LVGL display. However, this process should be completed by `bsp_display_lcd_init()` calling `lvgl_port_add_disp()`.

lznzhou33zzz, 2025-08-14:
Hello @Lzw655 

Regarding the code which you mentioned is not modified by myself, that is the origenal part from commit e59dbf74e882542df68eb8ebd4d061e895d270f3 , please check this commit to confirm whether it is correct. 
And what I am confused is why you mention lv_display_create(), it seems that there is no relation with the touch screen issue. 
Log is in ZIP file, would you mind to help me cross check?

closed 2025-08-14 by lznzhou33zzz