espressif/esp-brookesia #43: 打开SDKconfig错误

NicLam26, 2025-08-13:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

正常打开SDKconfig

### Actual behavior (suspected bug)

您好，最新更新的喵伴固件，在speaker下面打开，然后vs code打开SDK config，一直报下面的错误，说是Kconfig冲突，看起来是两个组件的Kconfig冲突，去掉其中一个还是不能解决问题，请问官方可以解决这个问题吗？

### Error logs or terminal output

```shell
-- Generating done (3.9s)

-- Build files have been written to: C:/Users/yukiz/Desktop/EchoEar/esp-brookesiav1.2/products/speaker/build

Running ninja in directory C:\Users\yukiz\Desktop\EchoEar\esp-brookesiav1.2\products\speaker\build
Executing "ninja confserver"...
[0/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\yukiz\Desktop\EchoEar\esp-brookesiav1.2\products\speaker\build && C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Scripts\python.exe C:/ESP32/Espressifv55/frameworks/esp-idf-v5.5/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file C:/Users/yukiz/Desktop/EchoEar/esp-brookesiav1.2/products/speaker/build/config.env && C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Scripts\python.exe -m kconfserver --env-file C:/Users/yukiz/Desktop/EchoEar/esp-brookesiav1.2/products/speaker/build/config.env --kconfig C:/ESP32/Espressifv55/frameworks/esp-idf-v5.5/Kconfig --sdkconfig-rename C:/ESP32/Espressifv55/frameworks/esp-idf-v5.5/sdkconfig.rename --config C:/Users/yukiz/Desktop/EchoEar/esp-brookesiav1.2/products/speaker/sdkconfig"

---------------------------ERROR--------------------------

info: INFO: Symbol ESP_IDF_VERSION defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:

    C:\Users\yukiz\Desktop\EchoEar\esp-brookesiav1.2\products\speaker\managed_components\espressif__esp_codec_dev\Kconfig:2

    C:\Users\yukiz\Desktop\EchoEar\esp-brookesiav1.2\products\speaker\managed_components\espressif__i2c_bus\Kconfig:4

-----------------------END OF ERROR-----------------------
Server running, waiting for requests on stdin...

---------------------------ERROR--------------------------

Traceback (most recent call last):
  File "asyncio\streams.py", line 545, in readline

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

  File "asyncio\streams.py", line 640, in readuntil
asyncio.exceptions.LimitOverrunError: Separator is found, but chunk is longer than limit

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf.py", line 871, in <module>
    main()
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf.py", line 761, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Lib\site-packages\click\core.py", line 1161, in __call__

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Lib\site-packages\click\core.py", line 1082, in main

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Lib\site-packages\click\core.py", line 1729, in invoke
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Lib\site-packages\click\core.py", line 1666, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\python_env\idf5.5_py3.11_env\Lib\site-packages\click\core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf.py", line 656, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf.py", line 214, in __call__

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    self.callback(self.name, context, global_args, **action_args)
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\core_ext.py", line 50, in build_target
    run_target(target_name, args, force_progression=GENERATORS[args.generator].get('force_progression', False))
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\tools.py", line 501, in run_target
    RunTool(generator_cmd[0], generator_cmd + [target_name], args.build_dir, env, custom_error_handler, hints=not args.no_hints,
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\tools.py", line 319, in __call__
    process, stderr_output_file, stdout_output_file = asyncio.run(self.run_command(self.args, env_copy))
                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "asyncio\runners.py", line 190, in run
  File "asyncio\runners.py", line 118, in run
  File "asyncio\base_events.py", line 653, in run_until_complete
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\tools.py", line 364, in run_command

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

    await asyncio.gather(
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\tools.py", line 443, in read_and_write_stream
    output = await read_stream()
             ^^^^^^^^^^^^^^^^^^^
  File "C:\ESP32\Espressifv55\frameworks\esp-idf-v5.5\tools\idf_py_actions\tools.py", line 399, in read_stream
    output_b = await input_stream.readline()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "asyncio\streams.py", line 554, in readline
ValueError: Separator is found, but chunk is longer than limit

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

Exception ignored in: <function BaseSubprocessTransport.__del__ at 0x00000201AC6EA8E0>
Traceback (most recent call last):
  File "asyncio\base_subprocess.py", line 125, in __del__
  File "asyncio\base_subprocess.py", line 78, in __repr__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

Exception ignored in: <function _ProactorBasePipeTransport.__del__ at 0x00000201AC70C040>
Traceback (most recent call last):
  File "asyncio\proactor_events.py", line 116, in __del__
  File "asyncio\proactor_events.py", line 80, in __repr__
  File "asyncio\windows_utils.py", line 102, in fileno
ValueError: I/O operation on closed pipe

-----------------------END OF ERROR-----------------------
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
-----------------------END OF ERROR-----------------------
```

### Steps to reproduce the behavior

用VS code打开SDKconfig就会报错

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows10

### Shell

ZSH

Lzw655, 2025-08-14:
Hi @NicLam26,

您好，感谢您的反馈，这个问题是因为 `i2c_bus` 和 `esp_codec_dev` 的 Kconfig 中具有重复的 `ESP_IDF_VERSION` 导致的，我们已经在 `i2c_bus` 组件中修复了这个问题，预计马上就能合入了。

NicLam26, 2025-08-14:
感谢，如果更新麻烦告知一下，谢谢！