espressif/esp-brookesia #47: 刷入固件后设备无限重启

xinxiushuo, 2025-08-22:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

1、PCB丝印为V1.1
2、修改代码编译没有报错，刷入设备后在开机logo哪里无限重启。
3、中控芯片用的ESP32-S3-WROOM-2-N32R8，考虑是不是这个芯片和出厂用的参数不同导致。

如果是的话如何调整，调整几个地方？

### Actual behavior (suspected bug)

1、PCB丝印为V1.1
2、修改代码编译没有报错，刷入设备后在开机logo哪里无限重启。
3、中控芯片用的ESP32-S3-WROOM-2-N32R8，考虑是不是这个芯片和出厂用的参数不同导致。

如果是的话如何调整，调整几个地方？

### Error logs or terminal output

```shell
--- Error: ClearCommError failed (PermissionError(13, '设备不识别此命令。', None, 22))
--- Waiting for the device to reconnect
I (837) esp_image: segment 1: paddr=003f1824 vaddr=3fc9f400 size=0a868h ( 43112) load
I (848) esp_image: segment 2: paddr=003fc094 vaddr=40374000 size=03f84h ( 16260) load
I (853) esp_image: segment 3: paddr=00400020 vaddr=42000020 size=32b414h (3322900) map
I (1632) esp_image: segment 4: paddr=0072b43c vaddr=40377f84 size=173ach ( 95148) load
I (1659) esp_image: segment 5: paddr=007427f0 vaddr=600fe000 size=00058h (    88) load
I (1671) boot: Loaded app from partition at offset 0xb0000
I (1671) boot: Disabling RNG early entropy source...
I (1673) MSPI Timing: Flash timing tuning index: 3
I (1673) octal_psram: vendor id    : 0x0d (AP)
I (1673) octal_psram: dev id       : 0x03 (generation 4)
I (1673) octal_psram: density      : 0x05 (128 Mbit)
I (1673) octal_psram: good-die     : 0x01 (Pass)
I (1674) octal_psram: Latency      : 0x01 (Fixed)
I (1674) octal_psram: VCC          : 0x00 (1.8V)
I (1674) octal_psram: SRF          : 0x01 (Fast Refresh)
I (1674) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (1674) octal_psram: BurstLen     : 0x01 (32 Byte)
I (1675) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (1675) octal_psram: DriveStrength: 0x00 (1/1)
--- Error: ClearCommError failed (PermissionError(13, '设备不识别此命令。', None, 22))
```

### Steps to reproduce the behavior

/

### Project release version

最新

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

win10

### Shell

ZSH

Lzw655, 2025-08-25:
Hi @xinxiushuo,

由于官方开发板使用的 PSRAM 是 16M 的，工程中默认开启了如下配置会占用约 7M 左右的 PSARM，使用 8M PSRAM 很可能会出现内存不足的问题，可以在 Menuconfig 中将这些配置关闭。

```
SPIRAM_XIP_FROM_PSRAM
SPIRAM_FETCH_INSTRUCTIONS
SPIRAM_RODATA
```

请注意，工程默认面向 PCB V1.2 使用，如使用 PCB V1.0/1.1，请在 Menuconfig 中选择 `BSP_PCB_VERSION_V1_2`。如果使用 VSCoe 插件打开 Menuconfig 出现报错，请参考 https://github.com/espressif/esp-brookesia/issues/48#issuecomment-3218170084 解决。

sunweigong, 2025-09-17:
> Hi [@xinxiushuo](https://github.com/xinxiushuo),
> 
> 由于官方开发板使用的 PSRAM 是 16M 的，工程中默认开启了如下配置会占用约 7M 左右的 PSARM，使用 8M PSRAM 很可能会出现内存不足的问题，可以在 Menuconfig 中将这些配置关闭。
> 
> ```
> SPIRAM_XIP_FROM_PSRAM
> SPIRAM_FETCH_INSTRUCTIONS
> SPIRAM_RODATA
> ```
> 
> 请注意，工程默认面向 PCB V1.2 使用，如使用 PCB V1.0/1.1，请在 Menuconfig 中选择 `BSP_PCB_VERSION_V1_2`。如果使用 VSCoe 插件打开 Menuconfig 出现报错，请参考 [#48 (comment)](https://github.com/espressif/esp-brookesia/issues/48#issuecomment-3218170084) 解决。

PCB V1.2似乎并没有开源（https://oshwhub.com/esp-college/echoear），当前在嘉立创能看到的PCB只是1.0（丝印为1.1）版本。如果项目支持PCB 1.2,请给予一下1.2pcb的开源地址，谢谢！

Lzw655, 2025-09-18:
> PCB [...]

Hi @sunweigong, PCB v1.2 仅为官方售卖版本，目前暂没有计划开源，请谅解。

luyongfugx, 2025-09-28:
> Hi [@xinxiushuo](https://github.com/xinxiushuo),
>
> 由于官方开发板使用的 PSRAM 是 16M 的，工程中默认开启了如下配置会占用约 7M 左右的 PSARM，使用 8M PSRAM 很可能会出现内存不足的问题，可以在 Menuconfig 中将这些配置关闭。
>
> ```
> SPIRAM_XIP_FROM_PSRAM [...]
这个问题貌似没解决。。。