espressif/esp-brookesia #51: 可以修改代码让speaker连接xiaozhi AI吗？

keroroyuki, 2025-08-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

因为原来固件的豆包coze每天对话很快就用完了，所以希望修改代码连接小智AI来暂时调试。但是看了代码不清楚修改哪里可以修改AI的连接。是否有方法可以修改呢？

Lzw655, 2025-08-26:
Hi @keroroyuki,

十分抱歉，当前还不能通过简单的修改实现连接 xiaozhi AI，不过内部正在进行相关功能的开发，后续应该是能够是实现的。

keroroyuki, 2025-08-26:
Hi@Lzw655 
那如果我想要连接的话，需要自己重新写websocket连接业务，以及语音的tts的逻辑吗？

dujianmin, 2025-08-27:
小智的代码底层驱动和整体逻辑和这个代码区别很大。要想移值的话。估计要重新按照小智的通讯协议架构了。难度很大。你们有没有对接小智大概的时间安排或者有没有可以连接小智的简单测试 demo 模仿一下

hyansuper, 2025-09-04:
这个是可以的，只要把做一个和 esp_coze 接口类似的小智组件。
我想我已经完成的大部分工作 https://github.com/hyansuper/xiaozhi_chat/  开源出来给有缘人继续完善 [doge]

大概思路是这样：
1. 喵伴[连接扣子对话的代码](https://github.com/espressif/esp-brookesia/tree/master/core/brookesia_core/ai_framework/agent)基本和 [esp-adf/coze_ws_app](https://github.com/espressif/esp-adf/tree/master/examples/ai_agent/coze_ws_app/) 这个例子相同，coze_ws_app 是一个单独的例子，相对简单，所以先跑通这一部分代码
2. 然后以 coze_ws_app 为参考，把通信协议部分换成小智的，见 [xiaozhi_chat/example](https://github.com/hyansuper/xiaozhi_chat/example）如果这个例子也能跑通
3. 那参考 coze_ws_app 与喵伴系统对接的方式，将其用 xiaozhi_chat 替换掉