espressif/esp-brookesia #54: 想自己定制speaker的屏幕的UI表情

keroroyuki, 2025-08-27:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

这个要如何操作呢？speaker的表情UI绘制是用到lvgl吗？具体是在哪实现的？

Lzw655, 2025-08-28:
Hi @keroroyuki,

表情 UI 绘制是基于 [image_player ](https://components.espressif.com/components/espressif2022/image_player/versions/1.1.1/readme) 组件绘制的，无需用到 lvgl，但是目前定制的灵活性上有一定限制，后续会更新更灵活的表情框架，可以先参考下面的步骤替换已有资源素材：

1. 要求：

* 源文件为 GIF 格式。
* 尺寸与工程里保持一致，emotion 为 `284x126`，icon 为 `64x64`。
* 文件名和数量与工程里保持一致，仅对已有素材进行替换，而非增减。

2. 生成 .aaf 文件：参考 image_player [README](https://github.com/espressif2022/image_player#scripts) 通过 python 脚本将 GIF 源文件转换为 .aaf 文件，以如下为例：

```c
// emotion
python <brookesia_path>/products/speaker/managed_components/espressif2022__image_player/script/gif_to_aaf.py <brookesia_path>/core/brookesia_core/docs/_static/assets/animations/emotion <brookesia_path>/core/brookesia_core/systems/speaker/assets/animations/emotion --split 126 --depth 8

// icon
python <brookesia_path>/products/speaker/managed_components/espressif2022__image_player/script/gif_to_aaf.py <brookesia_path>/core/brookesia_core/docs/_static/assets/animations/icon <brookesia_path>/core/brookesia_core/systems/speaker/assets/animations/icon--split 64 --depth 8
```

3. 重新编译烧录即可

keroroyuki, 2025-08-28:
那表情的gif文件的背景色，是需要透明还是需要纯黑？ @Lzw655

Lzw655, 2025-08-28:
> 那表情的gif文件的背景色，是需要透明还是需要纯黑？ [@Lzw655](https://github.com/Lzw655)

纯黑的。

Orlandolover2442-prog, 2025-11-26:
如果是使用的开机gif，那split是不是设置为360，为什么我的开机gif有点慢呢？感觉跟原图比慢了很多