espressif/esp-brookesia #57: coze_agent_config.c:26:16: error: implicit declaration of function 'strdup'  编译报错

myzhang001, 2025-09-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

环境：ubuntu22.04 
项目路径：esp-brookesia/products/speaker
编译报错

### Actual behavior (suspected bug)

编译报错

### Error logs or terminal output

```shell
coze_agent_config.c:26:16: error: implicit declaration of function 'strdup' [-Werror=implicit-function-declaration]
   26 |     return s ? strdup(s) : strdup("");
```

### Steps to reproduce the behavior

1.idf.build 

### Project release version

main bench

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu22.04

### Shell

ZSH

Lzw655, 2025-09-09:
Hi @myzhang001,

这个问题看起来是 *coze_agent_config.c* 中没有包含 `#include "string.h"` 导致的，能够麻烦加上试试，我会尽快修复这个问题，非常感谢！

myzhang001, 2025-09-09:
> Sent at: 2025年9月9日(Tue) AM10:21
>
>
> Hi @myzhang001,
>
> 这个问题看起来是 coze_agent_config.c 中没有包含 #include "string.h" 导致的，麻烦你加上试试，我会尽快修复这个问题，非常感谢！

好的，我更换了idf版本可以了