espressif/esp-brookesia #58: 请问喵伴底座固件会开源吗

WJ-JAY, 2025-09-06:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

请问喵伴底座固件会开源吗

dzlhk, 2025-10-26:
目前使用 EchoEar v1.0，底座通过磁吸座 UART 通信
可以提供最小协议说明：帧头、波特率、命令码（相对/绝对/停止）、长度字节规则、校验算法与覆盖范围、角度单位与缩放、大小端
若暂不便开源固件，是否可先公开协议或示例帧

dzlhk, 2025-10-26:
我想给予EchoEar（底座带按键可左右转动），实现开机自检可右转→左转→复位，或通过串口/应用指令精确控制角度，但项目内未见协议与用法文档，不知道咋实现？有大佬试过没有

Lzw655, 2025-10-27:
目前正在开发新版底座，之后软硬件都会开源。

dzlhk, 2025-11-09:
这个底座代码 ，能提供一下通讯协议也行哇！这样就可以调试了，或者新版本的代码预计什么时候发布呢！哈哈哈着急等待

dzlhk, 2025-11-14:
[image]这个代码指向那？