espressif/esp-brookesia #64: 没有SD卡，烧写源码后一直重启提示这个界面，正常应该10秒进入操作界面。

sunweigong, 2025-09-16:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

无SD卡的时候，正常应该提示10秒后进入操作界面。

### Actual behavior (suspected bug)

[image]

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

一个无SD的设备，烧录代码后会一直重启。

### Project release version

最新版本

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

windwos11

### Shell

PowerShell

sunweigong, 2025-09-16:
> 我也遇到了这个问题

解决了吗

Lzw655, 2025-09-16:
Hi @sunweigong,

我用官方的喵伴硬件以及最新的工程，发现并没有这个问题，请确认当前硬件为 16M PSRAM 模组，否则请参考 [issue](https://github.com/espressif/esp-brookesia/issues/47#issuecomment-3218599170)。

如果排除了硬件问题，请用串口助手抓取一下问题出现时的 LOG，谢谢！

sunweigong, 2025-09-17:
> 你好[@sunweigong](https://github.com/sunweigong)，
> 
> 我用官方的喵伴硬件以及最新的工程，发现并没有这个问题，请确认当前硬件为16M PSRAM模组，否则请参考[问题](https://github.com/espressif/esp-brookesia/issues/47#issuecomment-3218599170)。
> 
> 如果修复了硬件问题，请用串口助手提取一下问题出现时的LOG，谢谢！

问题应该是出现在这里。你们嘉立创开源的PCB版本是1.0，但实际上源码你们已经更新到了1.2。我在https://oshwhub.com/esp-college/echoear找了一下，发现你们似乎并没有开源1.1和1.2的PCB，因此在烧录之前需要退回重新选择一下支持版本，不知道我这么理解对不对？另外如果没有开源PCB1.1和1.2,源码应该予以注明的。

sunweigong, 2025-09-17:
复现方法：使用https://oshwhub.com/esp-college/echoear开源的硬件，然后git喵伴的固件源码，使用VSCODE打开，下方直接编译并烧录出现的次问题：

[image]

sunweigong, 2025-09-17:
我的ESP32用的是N32R16,是当前代码不适配的问题吗？

sunweigong, 2025-09-17:
PS D:\esp-brookesia\products\speaker> $env:IDF_PATH = 'C:\Espressif\v5.5\esp-idf';
PS D:\esp-brookesia\products\speaker>  & 'c:\Espressif\tools\python_env\idf5.5_py3.11_env\Scripts\python.exe' 'C:\Espressif\v5.5\esp-idf\tools\idf_monitor.py' -p COM17 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --make '''c:\Espressif\tools\python_env\idf5.5_py3.11_env\Scripts\python.exe'' ''C:\Espressif\v5.5\esp-idf\tools\idf.py''' --target esp32s3 'd:\esp-brookesia\products\speaker\build\brookesia_speaker.elf'
--- Warning: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM17 instead...
--- esp-idf-monitor 1.7.0 on \\.\COM17 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
I (827) esp_image: segment 1: paddr=003e4944 vaddr=3fc9eb00 sizeESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037f132
--- 0x4037f132: esp_cpu_wait_for_intr at C:/Espressif/v5.5/esp-idf/components/esp_hw_support/cpu.c:64
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2820,len:0x1710
load:0x403c8700,len:0xe5c
load:0x403cb700,len:0x31e0
entry 0x403c8940
I (33) boot: ESP-IDF v5.5 2nd stage bootloader
I (33) boot: compile time Sep 17 2025 17:08:46
I (33) boot: Multicore bootloader
I (34) boot: chip revision: v0.2
I (34) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (34) boot.esp32s3: SPI Mode       : SLOW READ
I (34) boot.esp32s3: SPI Flash Size : 16MB
I (35) boot: Enabling RNG early entropy source...
I (35) boot: Partition Table:
I (35) boot: ## Label            Usage          Type ST Offset   Length
I (35) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (36) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (36) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (37) boot:  3 model            Unknown data     01 82 00010000 00096000
I (37) boot:  4 factory          factory app      00 00 000b0000 007d0000
I (37) boot:  5 spiffs_data      Unknown data     01 82 00880000 000af000
I (38) boot:  6 anim_emotion     Unknown data     01 82 0092f000 001db000
I (38) boot:  7 anim_icon        Unknown data     01 82 00b0a000 00144000
I (39) boot:  8 anim_boot        Unknown data     01 82 00c4e000 000a0000
I (39) boot: End of partition table
I (39) boot: Defaulting to factory image
I (39) esp_image: segment 0: paddr=000b0020 vaddr=3c330020 size=33491ch (3361052) map
I (827) esp_image: segment 1: paddr=003e4944 vaddr=3fc9eb00 size=0a6a0h ( 42656) load
I (839) esp_image: segment 2: paddr=003eefec vaddr=40374000 size=0102ch (  4140) load
I (840) esp_image: segment 3: paddr=003f0020 vaddr=42000020 size=324df8h (3296760) map
I (1613) esp_image: segment 4: paddr=00714e20 vaddr=4037502c size=199f0h (104944) load
I (1642) esp_image: segment 5: paddr=0072e818 vaddr=600fe000 size=00058h (    88) load
I (1654) boot: Loaded app from partition at offset 0xb0000
I (1654) boot: Disabling RNG early entropy source...
I (1656) MSPI Timing: Flash timing tuning index: 2
I (1656) octal_psram: vendor id    : 0x0d (AP)
I (1656) octal_psram: dev id       : 0x03 (generation 4)
I (1656) octal_psram: density      : 0x05 (128 Mbit)
I (1657) octal_psram: good-die     : 0x01 (Pass)
I (1657) octal_psram: Latency      : 0x01 (Fixed)
I (1657) octal_psram: VCC          : 0x00 (1.8V)
I (1657) octal_psram: SRF          : 0x01 (Fast Refresh)
I (1657) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (1658) octal_psram: BurstLen     : 0x01 (32 Byte)
I (1658) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (1658) octal_psram: DriveStrength: 0x00 (1/1)

Lzw655, 2025-09-18:
这部分 LOG 仅为 bootloader 阶段打印的信息，无法看到问题相关信息，请问还有更多 LOG 吗？

sunweigong, 2025-09-19:
> 这部分 LOG 仅为 bootloader 阶段打印的信息，无法看到问题相关信息，请问还有更多 LOG 吗？

应该是没有错误，烧录都是正常的。但是就是不行，具体表现在插入SD卡不进界面一直重启，不插入SD就出现如题的问题。另外，烧录这个固件是没有问题的（https://ziliao.nologo.tech/d/%E6%97%A0%E5%90%8D%E7%A7%91%E6%8A%80Nologo%E8%B5%84%E6%96%99/ESP32%E5%BC%80%E5%8F%91%E6%9D%BF/%E5%96%B5%E4%BC%B4/%E5%9B%BA%E4%BB%B6/factory_echoear_speaker_esp32s3_release_v5_5.bin），但是官方的就是不行，包括我在嘉立创开源界面的（speaker_0_12_2_dev_echoear_1_0.bin）bin也试过了同样有问题。